netzero-labs/netzero #141: Automation for x 2 EV

cp23brown-del, 2025-08-09:
I have the following set up:

Solar, Tesla PW3, Zappi charger, Tesla Model 3 and Audi  Q4:

Can you please advise how I set automation up to charge both cars, the App currently is seeing my Tesla only, I would like to charge during the night as I’m on Octopus IOG and not drain PW, thankyou

zigam, 2025-08-09:
Netzero has a direct integration with IOG:
https://docs.netzero.energy/docs/tariffs/Intelligent-Octopus-Go

If you are charging both cars with IOG, you should not need any other automations. Is that the case?

cp23brown-del, 2025-08-09:
Thankyou, When charging it can also drain the battery and sometimes there seems to be conflict between the x 2 EV cars, I wanted to set something up where Netzero only allows charging after 11:30 and no drain from the battery

zigam, 2025-08-09:
If you want to only allow charging after 11:30pm, you'd have to first turn off IOG scheduled charging (since otherwise it might schedule charging outside of the off-peak window).

To prevent discharge of the Powerwall, you can create a pair of automations:
```
1. When vehicle charging starts: Set backup reserve to the current state of charge (preserve Powerwall charge).
2. When vehicle charging stops: Set backup reserve to: 20%.
```
(Replace 20% with your normal backup reserve.)

Then configure EV integrations. Go to Netzero `Settings > Vehicle Charging` and:

1. For the Tesla, pair the vehicle.
2. For the Zappi, set up the OCPP integration.

closed 2025-08-12 by zigam