netzero-labs/netzero #160: Prioritize Powerwall charging - Doesn't always work

philgrocks, 2025-09-27:
I have "Prioritize Powerwall charging" enabled.   With the Powerwall SOC set to 70% in Netzero.   My Powerwall is at 25% and I saw one of my cars is charging (I manually stopped it).

My cars and other info:
- Car 1 - 2023 Tesla model Y connected to Tesla Gen 3 Wall Connector
- Car 2 - 2024 Tesla model Y connected to another Tesla Gen 3 Wall Connector

Both cars have configured in the Tesla App to "Charge on Solar" and paired.

The feature of prioritize powerwall charging doesn't seem to work for me :-(

Thanks

philgrocks, 2025-09-27:
> zigam
>  left a comment
> (netzero-labs/netzero#160)
> Thanks for the report. Does the Netzero app indicate the car is charging when it starts charging? If there's an issue with the telemetry connection and Netzero is unaware of the up-to-date charging status, that can happen.

Yes the netzero app showed the car charging and I stopped the car charging from the netzero app.

Phil

zigam, 2025-09-27:
We’ll have to look at the data then, send us an email at feedback@netzero.energy